UbiquityRobotics/move_basic #24: Going backwards doesn't work

jim-v, 2017-09-21:
@jrlandau reported that goals behind don't work.

Question: should it handle this as a special case and actually move backwards, or do the arctan planner thing and rotate to face the goal and drive towards it?  If it moves backwards, then a front-facing LIDAR does not provide obstacle detection.

Definition of done:
Backwards goals result in behavior that considered reasonable by consensus.

jrlandau, 2017-09-21:
turning round takes quite some space, but moving backwards takes less, so
it may be useful to get out of a tight spot.

rohbotics, 2017-09-23:
Maybe it makes sense to go backward for short goals (<0.5 m) and do the atan2 method for backward goals beyond that distance?

rohbotics, 2017-09-23:
Decision:

If within a threshold backward and in angular threshold, then arctan backward.

Otherwise do the 'current behavior' sans bugs.

PR #25 (2017-09-24): Straight back
Closes #24
commit 25cc9831 (2017-09-24): Merge pull request #25 from UbiquityRobotics/straight-back

Closes #24
closed 2017-09-24 by jim-v