UbiquityRobotics/move_basic #32: Measured moves in robot commander

anfederman, 2018-02-13:
Prior to the January release measured moves ( turn right 90 degrees ) worked fine for both Loki and Magni,  
Even though no waypoints could be set. Now measured moves only work if aruco detect is running on Magni, and not at all on Loki.

Something is not working and I suspect a change to this node.

jim-v, 2018-02-13:
Can you run the command
```
rostopic list | grep move_base
```
After you give a move command and paste the output into this issue.

jim-v, 2018-02-13:
Also, please send the output of 
```
apt-cache policy ros-kinetic-rosbridge-server 
```
from the robots you have been testing on.

anfederman, 2018-02-13:
apt-cache policy ros-kinetic-rosbridge-server 
ros-kinetic-rosbridge-server:
  Installed: 0.8.6-0xenial-20180105-183806-0800
  Candidate: 0.8.6-0xenial-20180105-183806-0800
  Version table:
 *** 0.8.6-0xenial-20180105-183806-0800 500
        500 http://packages.ros.org/ros/ubuntu xenial/main armhf Packages
        100 /var/lib/dpkg/status

anfederman, 2018-02-13:
on boot. move_base topics:
 /rosbridge_ws/move_base/cancel
/rosbridge_ws/move_base/feedback
/rosbridge_ws/move_base/goal
/rosbridge_ws/move_base/result
/rosbridge_ws/move_base/status

now move basic or aruco connect so measured move commands in robot commander on loki don't work
after loading aruc and move_basic:

/move_base/cancel
/move_base/feedback
/move_base/goal
/move_base/result
/move_base/status
/move_base_simple/goal

I tried seeing if any topics were echoed while using the arrow keys, none were except status.

So I am not sure what's going on I also see:
/rosbridge_ws/move_base/cancel
/rosbridge_ws/move_base/feedback
/rosbridge_ws/move_base/goal
/rosbridge_ws/move_base/result
/rosbridge_ws/move_base/status

/tf2_web_republisher/cancel
/tf2_web_republisher/feedback
/tf2_web_republisher/goal
/tf2_web_republisher/result
/tf2_web_republisher/status
/tf_static

I was getting tf2 status, but none from rosbridge status.

I am guessing there are missing or incorrect transforms somewhere on Loki in the new loki_base

So arrow keys work, but measured move or set waypoint fails.

On RVIZ I gave a goal, and it tried to move:

status_list: 
  - 
    goal_id: 
      stamp: 
        secs: 1518545119
        nsecs: 574602031
      id: "/move_basic-1-1518545119.574602031"
    status: 1
    text: "This goal has been accepted by the simple action server"
---

So I think move-basic ain't talking to move base politely.

jim-v, 2018-02-13:
This is the problem:

anfederman, 2018-02-13:
/move_base/cancel
/move_base/feedback
/move_base/goal
/move_base/result
/move_base/status
/move_base_simple/goal
/tf
/tf2_web_republisher/cancel
/tf2_web_republisher/feedback
/tf2_web_republisher/goal
/tf2_web_republisher/result
/tf2_web_republisher/status

with rosbridge removed robot commander doesn't work at all.  Connects, but no movement even with arrow keys, though keyboard teleop works.  do we remap move_base to tf2???

rohbotics, 2018-02-13:
@anfederman Rosbridge is required for robot_commander to be able to do anything.

The issue is that rosbridge is using topic names without a leading slash, once we put rosbridge into it's own namspace to handle both ws and wss, this meant that the topics were also namespaced, breaking the code. 

@jrlandau can you make sure that all of the topic names in both robot commander and speech commands are fully qualified (aka start with a '/').

jrlandau, 2018-02-13:
@jrlandau <https://github.com/jrlandau> can you make sure that all of the
topic names in both robot commander and speech commands are fully qualified
(aka start with a '/').

I did that for the references to move_base, but it did not fix the problem
(measured move commands not working).

anfederman, 2018-02-14:
I just tested robot commander on Magni. Measured moves no longer work, neither do set waypoint or go to goal. RVIZ 2d nav go to goal does work. Prior to this weekend, robot commander was working. What happened after the huddle? This makes no sense, since I did not update or upgrade my Magni or Robot commander since last week.

jrlandau, 2018-02-14:
Do you do any launches when you bring the Magni up?

jim-v, 2018-02-14:
@rohbotics I'm fairly sure we've tested `Robot_Commander` since the rosbridges were moved into namespaces, so I think it's something that changed in `rosbridge_websocket` recently.  However, making all topics `Robot_Commander` uses absolute seems like it would make us immune to it.

jrlandau, 2018-02-14:
There are some things that may or may not be considered "topics" but which
need the same treatment.  Here is a function I had to change:

        function setWaypoint (waypointName) {
      //
----------------------------------------------------------------------
      // Sets a rosparam to contain the waypoint
      //
----------------------------------------------------------------------
            if (connected) {
                var waypoint = new ROSLIB.Param({
                    ros : ros,
                    name : "/waypoint/" + waypointName
                });
                function setWaypointParam (location) {
                    console.log ("Set waypoint " + waypoint.name + ": " +
location);
                    waypoint.set(location);
                    paramdump ();
                };
                // console.log ("getting the current pose");
                getPose (setWaypointParam);
            }
        }

jim-v, 2018-02-14:
@anfederman [tf2_web_repubisher](http://wiki.ros.org/tf2_web_republisher) also implements and actionlib interface, hence the topics:

```
/tf2_web_republisher/cancel
/tf2_web_republisher/feedback
/tf2_web_republisher/goal
/tf2_web_republisher/result
/tf2_web_republisher/status
```

They are not related to the namespace issue, which renames `move_base` topics to `/rosbridge_ws/move_base`.  As mentioned, the fix is for Robot_Commander to use `/move_base`.

jim-v, 2018-02-15:
@jrlandau sent me an updated APK file for Robot_Commander.

On the robot, which had `magni_base` service running, I ran

```
roslaunch magni_nav move_basic.launch 
```

Then on my phone, using the old app, I gave the verbal command "forward one meter".

The robot did not move and the bogus topics appeared:

```
rostopic list | grep move_base
/rosbridge_ws/move_base/cancel
/rosbridge_ws/move_base/feedback
/rosbridge_ws/move_base/goal
/rosbridge_ws/move_base/result
/rosbridge_ws/move_base/status
```

Then, I installed the new app on my phone and restarted everything on the robot, to be in a clean state:
```
sudo service magni-base stop
sudo service magni-base start
roslaunch magni_nav move_basic.launch 
```

Once again, I gave the verbal command "forward one meter".

This time, the bogus topics did not appear, and move_basic reported receiving the goal:

```
[ INFO] [1518674933.957489772]: Received goal 1.000000 0.000000 0.000000 base_link
```
The robot moved forward.

I am closing this issue.  If there are issues with localization, they should be opened as new issues.

closed 2018-02-15 by jim-v