UbiquityRobotics/move_basic #36: Linker problem at Intel Euclid / Ubuntu 16.04

mockbot, 2018-03-01:
Hi,
I'm trying to compile move_basic on Intel Euclid with Ubuntu 16.04 (ros kinetic).
During catkin_make process, I get a linker error:
[  0%] Linking CXX executable /intel/euclid/euclid_ws/devel/lib/move_basic/move_basic
CMakeFiles/move_basic.dir/src/obstacle_detector.cpp.o: In function `ObstacleDetector::range_callback(boost::shared_ptr<sensor_msgs::Range_<std::allocator<void> > const> const&)':
obstacle_detector.cpp:(.text+0xd1c): undefined reference to `void tf2::fromMsg<geometry_msgs::Point_<std::allocator<void> >, tf2::Vector3>(geometry_msgs::Point_<std::allocator<void> > const&, tf2::Vector3&)'
obstacle_detector.cpp:(.text+0xfa4): undefined reference to `void tf2::fromMsg<geometry_msgs::Vector3_<std::allocator<void> >, tf2::Vector3>(geometry_msgs::Vector3_<std::allocator<void> > const&, tf2::Vector3&)'
obstacle_detector.cpp:(.text+0x102b): undefined reference to `void tf2::fromMsg<geometry_msgs::Vector3_<std::allocator<void> >, tf2::Vector3>(geometry_msgs::Vector3_<std::allocator<void> > const&, tf2::Vector3&)'
CMakeFiles/move_basic.dir/src/obstacle_detector.cpp.o: In function `ObstacleDetector::scan_callback(boost::shared_ptr<sensor_msgs::LaserScan_<std::allocator<void> > const> const&)':
obstacle_detector.cpp:(.text+0x188d): undefined reference to `void tf2::fromMsg<geometry_msgs::Point_<std::allocator<void> >, tf2::Vector3>(geometry_msgs::Point_<std::allocator<void> > const&, tf2::Vector3&)'
obstacle_detector.cpp:(.text+0x1917): undefined reference to `void tf2::fromMsg<geometry_msgs::Vector3_<std::allocator<void> >, tf2::Vector3>(geometry_msgs::Vector3_<std::allocator<void> > const&, tf2::Vector3&)'
collect2: error: ld returned 1 exit status
move_basic/CMakeFiles/move_basic.dir/build.make:348: recipe for target '/intel/euclid/euclid_ws/devel/lib/move_basic/move_basic' failed
make[3]: *** [/intel/euclid/euclid_ws/devel/lib/move_basic/move_basic] Error 1
CMakeFiles/Makefile2:6716: recipe for target 'move_basic/CMakeFiles/move_basic.dir/all' failed
make[2]: *** [move_basic/CMakeFiles/move_basic.dir/all] Error 2
CMakeFiles/Makefile2:6728: recipe for target 'move_basic/CMakeFiles/move_basic.dir/rule' failed
make[1]: *** [move_basic/CMakeFiles/move_basic.dir/rule] Error 2
Makefile:2526: recipe for target 'move_basic' failed
make: *** [move_basic] Error 2
Invoking "make move_basic -j4 -l4" failed

I've installed all tf2 related libs.
Any idea, which library is missing at my system ?

Thanks in advance
Chrimo

jim-v, 2018-03-01:
It seems that
```
void fromMsg(const geometry_msgs::Vector3& in, tf2::Vector3& out)
```
is not a library function, but is implemented in
 `/opt/ros/kinetic/include/tf2_geometry_msgs/tf2_geometry_msgs.h`

Can you try adding

```
#include <tf2_geometry_msgs/tf2_geometry_msgs.h>
```
to `move_basic/src/obstacle_detector.cpp` and let me know what happens.

mockbot, 2018-03-02:
I've tried it with your suggestions:

but

please watch this:

Scanning dependencies of target move_basic
[  0%] Building CXX object move_basic/CMakeFiles/move_basic.dir/src/obstacle_detector.cpp.o
[  0%] Linking CXX executable /intel/euclid/euclid_ws/devel/lib/move_basic/move_basic
CMakeFiles/move_basic.dir/src/obstacle_detector.cpp.o: In function `ObstacleDetector::range_callback(boost::shared_ptr<sensor_msgs::Range_<std::allocator<void> > const> const&)':
obstacle_detector.cpp:(.text+0xd1c): undefined reference to `void tf2::fromMsg<geometry_msgs::Point_<std::allocator<void> >, tf2::Vector3>(geometry_msgs::Point_<std::allocator<void> > const&, tf2::Vector3&)'
obstacle_detector.cpp:(.text+0xfa4): undefined reference to `void tf2::fromMsg<geometry_msgs::Vector3_<std::allocator<void> >, tf2::Vector3>(geometry_msgs::Vector3_<std::allocator<void> > const&, tf2::Vector3&)'
obstacle_detector.cpp:(.text+0x102b): undefined reference to `void tf2::fromMsg<geometry_msgs::Vector3_<std::allocator<void> >, tf2::Vector3>(geometry_msgs::Vector3_<std::allocator<void> > const&, tf2::Vector3&)'
CMakeFiles/move_basic.dir/src/obstacle_detector.cpp.o: In function `ObstacleDetector::scan_callback(boost::shared_ptr<sensor_msgs::LaserScan_<std::allocator<void> > const> const&)':
obstacle_detector.cpp:(.text+0x188d): undefined reference to `void tf2::fromMsg<geometry_msgs::Point_<std::allocator<void> >, tf2::Vector3>(geometry_msgs::Point_<std::allocator<void> > const&, tf2::Vector3&)'
obstacle_detector.cpp:(.text+0x1917): undefined reference to `void tf2::fromMsg<geometry_msgs::Vector3_<std::allocator<void> >, tf2::Vector3>(geometry_msgs::Vector3_<std::allocator<void> > const&, tf2::Vector3&)'
collect2: error: ld returned 1 exit status
move_basic/CMakeFiles/move_basic.dir/build.make:348: recipe for target '/intel/euclid/euclid_ws/devel/lib/move_basic/move_basic' failed
make[3]: *** [/intel/euclid/euclid_ws/devel/lib/move_basic/move_basic] Error 1
CMakeFiles/Makefile2:6716: recipe for target 'move_basic/CMakeFiles/move_basic.dir/all' failed
make[2]: *** [move_basic/CMakeFiles/move_basic.dir/all] Error 2
CMakeFiles/Makefile2:6728: recipe for target 'move_basic/CMakeFiles/move_basic.dir/rule' failed
make[1]: *** [move_basic/CMakeFiles/move_basic.dir/rule] Error 2
Makefile:2526: recipe for target 'move_basic' failed
make: *** [move_basic] Error 2
Invoking "make move_basic clean -j4 -l4" failed

trainman419, 2018-03-02:
Cross-post: https://answers.ros.org/question/284071/move_basic-linker-problem-intel-euclid-ubuntu-1604/

jim-v, 2018-03-04:
@mockbot We have been unable to come up with a solution to your problem.  The package builds for us on X86_64, Raspberry Pi, and in the ROS build farm.  Do you have problems building any other packages?

mockbot, 2018-03-16:
ReHi

yes, I've successfully compiled rtabmap and some other small PKGs with catkin_make.

Hint: at this Intel Euclid System I cannot compile  dnn_detect, too.
Possible a problem with OpenCV version ?
This Intel beast seems not to be a normal X86_64 ;-( 

Thanks for your friendly response
Cheers
Chrimo

rohbotics, 2018-09-22:
Closing due to age and weird platform bugs.

Hope it all worked out well

closed 2018-09-22 by rohbotics