UbiquityRobotics/move_basic #72: move_basic does not respect max_angular_velocity parameter

JanezCim, 2020-11-12:
ROS: kinetic

After move_basic is launched, the param max_angular_velocity is confirmed to be set to 0.01rad/s:
```
ubuntu@housie ~ $ rosparam get /move_basic/max_angular_velocity
0.009999999776482582

```
but when the robot is moving, im seeing big jumps in cmd_vel topic:

JanezCim, 2020-11-13:
Found out that parameter `max_angular_velocity` is meant for maximum angular velocity, while robot is **turning on the spot**, while the parameter `max_lateral_rotation` controls the maximum angular velocity while **driving towards goal**. This answers this issue but maybe the names of the parameters are a bit unintuitive? @rohbotics  

**suggestion:** max_turning_velocity, max_lateral_velocity
**suggestion2:** we set the max_lateral_velocity = max_turning_velocity as default with 
`nh.param<double>("max_lateral_velocity", maxLateralVelocity, maxTurningVelocity);`
as this would still lave the user the control over both, but default parameters would be a bit easier to understand.

tp4348, 2020-11-13:
I would go with the suggestion(1), because it kind makes sense for me that one stands for the velocity when turning in-place, while the other stands for the lateral velocity, while robot should be moving linear. I can imagine the cases where these two should have different default values.

JanezCim, 2020-11-25:
So I'm going to close this issue because we found the answer to the main question. Im going to open a new one for the unintuitive parameters-

closed 2020-11-25 by JanezCim