UbiquityRobotics/move_basic #73: remaining distance stays the same

manosmaroulis, 2020-11-20:
Hello i am trying to use the move_basic package for a simple inertial navigation system,i provide odometry and imu data to the robot_pose_ekf package and publish my transform message from odom to base_link. I use driving in odom frame and tried planning in both odom and base_link frame, but when i use the odom planning frame my rovers seems to do a circle and stop and when i try base_link the remaining distance stays still and the rover never stops running.Thank you in advance.

tp4348, 2020-12-08:
#63 should solve that. If it still doesn't solve the problem for you, please re-install the move_basic pkg. Last but not least record the rosbag and post it here so we can debug it.

tp4348, 2021-02-08:
Re-open this issue if it still present on your platform.

closed 2021-02-08 by tp4348