UbiquityRobotics/move_basic #78: Add option to not rotate when the received quaternion is not unit

MoffKalast, 2020-12-22:
We've discussed this a fair bit, ultimately it would be of great help in many projects if move basic could just be told to go to a goal with no proper rotation specified. The most straightforward way of doing this seems to be just sending an invalid rotation, which does intuitively make sense and goes into the realm of expected behaviour.

Optionally we may need to add a parameter to enable this, but given the other changes to move_basic lately we're likely breaking most people's setup anyhow so it's not a high priority.

JanezCim, 2021-01-11:
This was addressed in 335df40cd4a62d65060571c538fb789a47150731 and is not being tested in aislemode. Will not be added to main branch until the relevant people agree

MoffKalast, 2021-01-11:
> until the relevant people agree

[image]

tp4348, 2021-03-22:
Done: https://github.com/UbiquityRobotics/move_basic/blob/f855e697873ede218f482488ddfff3125bd40844/src/move_basic.cpp#L380

closed 2021-03-22 by tp4348