UbiquityRobotics/move_basic #80: External force during goal following

tp4348, 2021-02-08:
Currently if the robot is moving towards the goal and is moved or hit by some other dynamic object it does not update it's pose according to the goal and therefore misses it.

PROPOSED SOLUTION: 
Keep track of goal_frame~base_frame transform during movement and update it accordingly if needed.

MoffKalast, 2021-04-22:
@tp4348 I've just done a round of tests with ezmap and feature-external-force and it performs super well, at least as long as the map localization manages to actually find the new position properly. So if it works well in real life tests I'd vote for it to be merged into kinetic-devel as soon as possible since it's really a much needed addition.

tp4348, 2021-04-23:
Well yeah it's updating in map frame haha. I think I will be able to test it on Sunday if @GreTimotej can't make it earlier.

MoffKalast, 2021-04-23:
Yep, it's expected, I'm just saying we might need to iris_lama next if we're still having problems afterwards.

GreTimotej, 2021-07-06:
Today I tried the above mentioned issue. I was moving the robot while driving and it went back to its course. I think that is it?

MoffKalast, 2021-07-06:
That's mostly it, but I think there were some issues with it the first time it was tested? Are those resolved?

tp4348, 2021-07-07:
Timotej was the one testing it so I guess since the second testing was successful we can close this.

tp4348, 2021-07-07:
On which branch were you testing @GreTimotej?

closed 2021-08-30 by tp4348