UbiquityRobotics/move_basic #83: Terminal status for goals

MoffKalast, 2021-02-18:
It seems that occasionally when using move_basic through the action client it's possible to get the following result when preempting:

```
[ WARN] [1613663891.808006341, 985.273000000]: Your executeCallback did not set the goal to a terminal status.
This is a bug in your ActionServer implementation. Fix your code!
For now, the ActionServer will set this goal to aborted

```

Upon which move_basic seems to lock up sometimes. The solution would be to check in which case it's possible to end the callback without calling set_something for the goal to anything before returning. Found in  nav-features-aislemode branch, possibly others.

MoffKalast, 2021-02-18:
After more research it seems to be a bug in the action client itself, so we'll have to watch to never get into situations where this can occur instead.

https://github.com/ros/actionlib/issues/126

closed 2021-02-18 by MoffKalast

tp4348, 2021-02-19:
From my experience this is usually not a bug in the actionServer but rather our code that set a goal two time successful or aborting the same goal multiple times.

MoffKalast, 2021-02-19:
Well in this case it may just be both 😑