UbiquityRobotics/move_basic #86: Using both sonars and lidar at the same time for collision detection

MoffKalast, 2021-03-02:
In cases where the lidar is mounted super high or low on the robot we can't detect all obstacles and must also look at the sonars and not just the laserscan despite it being available.

tp4348, 2021-03-02:
Both the sonar and LiDAR data should be published at the same time. I guess the frequency of the sonar data is to low or something since:

https://github.com/UbiquityRobotics/move_basic/blob/7c277c3270a00cead60f65f3c883a2633905ceb0/src/obstacle_points.cpp#L204

I will test this and post results here.

tp4348, 2021-04-08:
I've tested on Magni and points (detections) from both are used. I think this might be more of a Polaris problem. 
@MoffKalast  can we close this issue or should wait for Polaris to arrive in TP in 2022 ;)?

MoffKalast, 2021-04-09:
I've tested it myself a while back and yes it should be working in theory, but polaris doesn't really stop using sonars in gazebo either.

Right now I'm thinking of doing a separate node that blocks movement once only the front sonar gets blocked for a specified time, so I can tweak it specifically for these trash sensors. Gonna be a bit hard to do without the actual hardware but I think mark has a few he could potentially test with.

tp4348, 2021-04-19:
I think this should be easily feasible after #90 closes.

tp4348, 2021-07-19:
Confirmed working using both Sensor and LiDAR. #90 is a separate long-term issue.

closed 2021-07-19 by tp4348