UbiquityRobotics/move_basic #89: Aborting a goal due to moving away from goal needs to be ROS option enabled

mjstn, 2021-03-26:
We recently went through a great deal of debug time tracking down why Aisle mode was aborting after turning around at the end of an aisle.    We discovered it was due to 'Moving away from goal'.
Aisle mode has its own logic above move basic to deal with abort of goal.  too many cooks spoil the broth.

Aisle mode will need that sort of abort to be disabled and I suggest ROS option.

tp4348, 2021-03-27:
@JanezCim this can be implemented in the "side-branch" for Aisle mode, but at the same time should also be debugged what is causing it and fix it for master branch.

JanezCim, 2021-03-29:
Maybe lets not have any more side branches :) I think the smartest way may be that we go through the changes together with @tp4348 and see what might be causing the problems.