UbiquityRobotics/move_basic #91: Suggest we use laser_filters instead of scan filtering in obstacle_points.cpp

tp4348, 2021-04-09:
There is a lot going on in obstacle_points.cpp to filter LiDAR scan to get the actual obstacle points.

I would suggest to use laser_filters ROS package which has a support for Kinetic, Melodic, Noetic. More here:  http://wiki.ros.org/laser_filters

The only thing that bothers me is that this then mean that move_basic is not a standalone package anymore but rather uses another ROS package.

What do you think @mjstn, @JanezCim, @MoffKalast, @rohbotics ?

MoffKalast, 2021-04-09:
Well all of our actual software suites already use the laser filters, so that would make sense. 

Unfortunately move_basic also needs to work in a separate fashion so I'd suggest making a launch file that demoes a laserscan setup by launching move_basic, remapping scan to scan_filtered and launching the filters too.

JanezCim, 2021-04-14:
I agree with having laser filters instead because they include some super useful features like shadow filtering and such if we need them down the road. It can get pretty CPU intensive for PI tho, so i would always be concious of that. You can find an example of what @MoffKalast suggest in other project repos :)