UbiquityRobotics/move_basic #94: Rotation-only goals ignored

MoffKalast, 2021-06-07:
So it would seem that sending simple or actionclient goals without a translational change results in a complete disregard of the goal.

For example, sending a simple goal:

```
[ INFO] [1623088413.784146154, 48.450000000]: MoveBasic: Received simple goal
[ INFO] [1623088413.869282423, 48.500000000]: MoveBasic: Received goal 0.000000 0.000000 180.000000 base_link
[ INFO] [1623088413.869342759, 48.500000000]: Planning in goal frame: base_link

[ INFO] [1623088413.869401961, 48.500000000]: MoveBasic: Goal in base_link  0.000000 0.000000 180.000000
[ INFO] [1623088413.869471346, 48.500000000]: MoveBasic: Goal in base_link  0.000000 0.000000 180.000000
```

The base proceeds to do nothing. It seems that the threshold for movement is at around 4cm on both x and y, so if we send a goal at 0.04 0.04 180 it does actually work. Anyhow, pretty annoying and I'm almost sure this used to work at some point.

The tf frame doesn't seem to matter, it also ignores ones in map and odom if the translation sent matches the one of the robot. Steps to reproduce would be to load up rviz and send a manual goal in a 8x8cm area around the robot center.

MoffKalast, 2021-06-07:
Another thing, the workaround I mentioned doesn't fully work either it seems.

```
[ INFO] [1623089989.156225937, 814.851000000]: Planning in goal frame: base_link

[ INFO] [1623089989.156259677, 814.851000000]: MoveBasic: Goal in base_link  0.040000 0.040000 180.000000
[ INFO] [1623089989.156328205, 814.851000000]: MoveBasic: Goal in base_link  0.040000 0.040000 180.000000
[ INFO] [1623089989.156369680, 814.851000000]: MoveBasic: Requested rotation -129.398652
[ INFO] [1623089993.879035292, 817.175000000]: MoveBasic: Done rotation, error 3.706254 degrees
[ INFO] [1623089994.417629743, 817.397000000]: MoveBasic: Done linear, error: x: 0.042650 meters, y: 0.004839 meters
[ INFO] [1623089994.418079954, 817.397000000]: MoveBasic: Requested rotation 2.368312
[ INFO] [1623090006.863490802, 823.137000000]: MoveBasic: Done rotation, error 4.481289 degrees
```

The rotation and translation do get executed, but the goal itself doesn't get labeled as succeeded, but stays in the active state (1), so the actionclient doesn't know it's done and cannot proceed.

(this is using the latest kinetic-devel, but the same thing happens on feature-externalforce so it's bound to be a somewhat more widespread issue)

tp4348, 2021-06-09:
Done in 629c0815c25586fe0e125247dd37b5224a89b7a6

closed 2021-06-09 by tp4348

tp4348, 2021-11-04:
I cannot reproduce the issue in Gazebo, it works for me. I was publishing rotation only goals with **rostopic pub**.

MoffKalast, 2021-11-04:
I think I managed to reproduce it with rviz simple goals, just dropped on the robot. If that's the case and you're running it with rosrun it may be some parameter changed by ezmap's launch of move_basic that causes this.