UbiquityRobotics/move_basic #101: More aggressive external force correction

MoffKalast, 2021-11-02:
In the stock configuration it seems that the robot doesn't quite correct its drift quickly enough if the goal is a close distance (<2m or so) away. There is probably a curve somewhere or a set of PIDs that determine how much the robot steers towards the right location. That part seems to need some boost when the offset isn't much as it doesn't get corrected enough currently.

tp4348, 2021-11-04:
Can we get a video of that? How much was the robot moved of coarse? Was it done intentially for testing purposes or can mentioned external force actually happens more often that not in the clients scenario?

Yes, the PID configuration is rather soft(by default a bit slower corrections) and can be retuned but there is no "best" general parameters, since the PIDs also highly affect the driving perfomance with the payload etc. 
we could potentially try something like PIDV, otherwise I think we should strive to have some kind of auto tuner or tune the PIDs according to client needs.

MoffKalast, 2021-11-04:
Well I've been shown some live demos via zoom, but I don't have any recordings to show. It's a thing that's either caused by lidar jumps or the hardwood floor they're using I would guess.

There are several other factors, but the main one is the robot moving sideways on the map while driving due to very minor lidar rotational misalignment. So in effect you get lidar jumps that don't get corrected fast enough.

I've attempted to increase the lateral pids, which seems to somewhat improve the situation, but it's hard to actually get the desired behaviour. It's not gonna be an issue with gen 6 as it's a driver level auto tuner by design but we're stuck with this for the time being I guess.