UbiquityRobotics/move_basic #99: Sending goal succeeded when it doesn't seem to have

MoffKalast, 2021-11-02:
Something odd reported by a client, they seem to be getting a "MoveBasic: Done linear" in cases where the robot hits the goal, but more often than not the case is as follows:

- nothing gets printed in the console
- linear is done, but not very accurately
- final rotation is done
- action client receives goal succeeded despite being more than linear_tolerance away from goal, again nothing printed in the console

It may just be the extras added in the runaway-rotation-fix branch they're currently on, but it's worth a closer look as to which code path ends this way.

tp4348, 2021-11-04:
Seems to me that driving in map frame imposes a whole lots of problems we don't yet understand.
Move_basic was extensively tested using odometry (driving odom frame) but as soon as we integrated it with EzMap and LaMa bunch of I think localization problems is accuring that affect the navigation perfomance. 

Can the client try and see the perfomance using odom only, without map. This can help us understand better where the problem is. Ofcourse during drive in odom, external-force applied or wheel slip will mess up the robot final destination but I think this is not the crucial feature for the mentioned client.

tp4348, 2021-11-04:
Nothing gets printed in the console?
Linear goal is done? 

When, what we they trying to achieve and failed?

MoffKalast, 2021-11-04:
> Can the client try and see the performance using odom only, without map

Hmm interesting idea, it would require a slight modification of sending the goals in the odom frame instead of map, but worth a shot to see what exactly happens. But probably nothing too useful I'd imagine.

> Nothing gets printed in the console?

Apparently. I think they filtered out the "remaining x" messages, but nothing in terms of goal finishing somehow.

> Final rotation is done?

Yep.

> When, what we they trying to achieve and failed?

Just normal route navigation with ezmap, nothing out of the ordinary actually. They've been having issues with this since the start basically.

MoffKalast, 2021-11-04:
> see the perfomance using odom only, without map

So, a follow up on that, I've attempted to set all frames to odom which I think forces internal transform of the goal into odom and then working with that iirc.

    <param name="preferred_planning_frame" value="odom"/>
    <param name="alternate_planning_frame" value="odom"/> 
    <param name="preferred_driving_frame" value="odom"/>
    <param name="alternate_driving_frame" value="odom"/> 

The results were mostly the same, at least for these 1m distance goals I can test with.