UbiquityRobotics/move_basic #107: RootRework: Better navigation?

MoffKalast, 2022-03-22:
As discussed we'll try to implement two types of path planning:

- A* paths to do collision avoidance (Janez)
- a system that doesn't use occupancy grids (Jan)

JanezCim, 2022-03-24:
OK with a* this is where i am now:
[image]

in Gazebo you set a nav goal, ros navigation global planner calculates it using a*, outputs goal and magni follows it. 
Next planned work: recaculate that path every x meters/ y seconds, recalculate goal if obstacle on path

MoffKalast, 2022-03-24:
> and magni follows it using move_basic

Wait, does move basic actually have path following capabilities?

JanezCim, 2022-03-24:
Nope, goal manager is passing the specific points through actions

MoffKalast, 2022-03-24:
Ah ok, so how does it determine which points to send? Something like the farthest one in direct view?

MoffKalast, 2022-03-25:
Hmm I wonder if we could instead do this with sort of local map that updates faster. Instead of relying on a slam node, we could instead take the laserscan and treat any detections as obstacles to avoid and the rest as open space. There has been a lot of demand for very fast driving so that probably be considered here.

JanezCim, 2022-03-28:
Im not sure that raspberry would be up to the task but we can try

JanezCim, 2022-03-28:
Teb local planner in action:

[image]

JanezCim, 2022-04-20:
@JanezCim tries out a* on real robot