UbiquityRobotics/move_basic #109: Unable to set up reversing API

Connor-woodahl, 2022-09-14:
When trying to run the following on our AGV, 

cd /home/ubuntu/catkin_ws/src/move_basic
git pull
git checkout reversing
cd /home/ubuntu/catkin_ws/src/ezmap
git pull
git checkout v1.1.4-QCM-reversing

I ran into an error when trying to perform the second pull: 
remote: Repository not found.
fatal: repository 'https://github.com/UbiquityRobotics/ezmap.git/' not found

anfederman, 2022-09-14:
Ez map is a private repository, you need to be given permission to see or
download.

anfederman, 2022-09-14:
Dave Crawley  ***@***.***) Is an owner, and can grant
permission to ezmap. You will need to tell us your github account name.

Connor-woodahl, 2022-09-14:
Github account: Connor-woodahl

MoffKalast, 2022-09-15:
Seems like your initial email invite has expired, check your inbox for another one.

closed 2022-09-15 by Connor-woodahl